Weber-State-Submarine-Project/Submarine #25: Begin testing Sonar Sensor

ZefraSK, 2024-05-31:
DONE: When readable data is achieved from Sonar

CONSIDER:
-  Seeing data in RVIZ
-  Find actual max value
-  Determine value when can't see wall

Ben1219925, 2024-08-02:
The Sonar works pretty accurately within +-5 mm at small ranges, haven't tested large ranges yet.
Note that in the dfrobot documentation the arduino wiring diagram/code is correct but the board overview RX and TX are switched the wiring should actually be as follows:
Red line 	        VCC 	power supply input positive pole
Black line 	GND 	power ground wire
white line 	RX 	        UART receiving data line
yellow line 	TX 	        UART transmitting data line

The Arduino serial and calculation code can be almost directly translated to python, which will work with ros2 on the rpi, however in the python code when calculating the check sum it must be modded by 256 to stay in the 0-255 range, which wasn't necessary in the original Arduino code.

Need to find the actual max range still and test at larger distances, current test cases have been 0-0.16 m should test up to 6 m. Waterproofing seems to be holding well.

Ben1219925, 2024-08-09:
This week:
Test accuracy at larger distances
Determine +- mm reliablity

Future:
Determine actual max distance
Test for echo/interference in echo prone environments such as the corner of a pool
Test for value given when no objects are in measuring distance.

closed 2024-08-30 by ZefraSK