Weber-State-Submarine-Project/Submarine #27: Display map in WebUI

ZefraSK, 2024-05-31:
DONE: Displays the map on the web

ZefraSK, 2024-06-28:
Everything they wrote is for ROS1, really annoying to import for ROS2. Currently looking into other options. -Ben

Ben1219925, 2024-07-13:
Consider using Foxglove to view the map. May also be possible to include in our own website as an iframe if it works

[https://docs.ros.org/en/jazzy/How-To-Guides/Visualizing-ROS-2-Data-With-Foxglove-Studio.html](https://docs.ros.org/en/jazzy/How-To-Guides/Visualizing-ROS-2-Data-With-Foxglove-Studio.html)

[https://app.foxglove.dev/weber-state-submarine-project/dashboard](https://app.foxglove.dev/weber-state-submarine-project/dashboard)

Ben1219925, 2024-07-27:
Foxglove seems to function like a web version of rviz, plus a lot of other cool features that we won't need.

We could just use foxglove instead of our own custom website since it can already view 3d objects live just like rviz, it can publish messages at the push of a button such as our start stop button, and it can display messages such as the log I made with our custom site.

If we go with just foxglove we will likely need to adjust the wordage of our deliverables so that instead of creating a webUI we use Foxglove to do the same tasks.

The primary reason for possibly keeping the custom site is to show that we did more work and engineering, instead of just using a tool like Foxglove, it will likely seem more technically impressive to build something ourselves.

If desired I think I can incorporate Foxgloves 3d pane into our existing custom website using an iframe or something similar, will investigate.

closed 2024-08-09 by Ben1219925