Weber-State-Submarine-Project/Submarine #42: Test Speed Sensor

ZefraSK, 2024-06-28:


ZefraSK, 2024-08-09:
Handing off to ben Tuesday or Thursday

Ben1219925, 2024-08-16:
The speed sensor seems to work as intended on when tested on the arduino.
The provided code has a few logical flaws that will need to be fixed but it will be an easy code conversion to python.

We may need an ADC for it to work properly with the raspberry pi, I will test it using the UART to USB and see if we need one.

The value alternates between 0 (sometimes 1) and 1023 (1020-1023) when at 5v power supply.

TODO: 
Convert arduino code to Ros2 Python node with odom message type
Test with UART to USB to see if an ADC is necessary
Test speed accuracy
Clean up wiring to remove breadboard
Consider how to mount to the submarine

Ben1219925, 2024-08-16:
Using my Arduino as a standin ADC I was able to receive the signal and relay the message to my laptop using UART, so if we get an ADC it should work properly.

ZefraSK, 2024-08-16:
If we cannot use an arduino, one option could be [ADS112U04](https://www.ti.com/product/ADS112U04?utm_source=google&utm_medium=cpc&utm_campaign=asc-null-null-gpn_en-cpc-pf-google-wwe&utm_content=ads112u04&ds_k=ADS112U04&dcm=yes&gad_source=1&bm-verify=AAQAAAAJ_____5b8ETHFwDJp1bZxrR7NTK292Pdaoe7zZflBtJHAQHU7QsPxigrIaz88d6GegHooL9tYUe-ZW4bmQvKqQYk15dyOk-NPfhGLLnXlSdhEAU6Hs2TD0lcgDCQbTlkfq0oWV0uG1lBUCrtNLHfca1oq9sCf7aGtKZE6Of0ELF5f8103Hqm5Gm8NVpK3mLoTgSqXoBxe0vF-mtwOjdGwbshX8nARfeAMaGX9UF-GFgX2Iz_tyLGbc9h4d8sJfvVTRDI85GdAqgvQoinv7TeOjxnlaMG4vXASfcy244leyEA2hkZEMSwhSgIqY-ddwh8Av2SdmjjRPErF3TOl8XwNoVNDD65jOAzojJ3bkRxFHuflj5DuyED9NeL6jmzIS1NJXnQ28wCh7SgG4cKX0SfRHiGkMvfGmUHTHMq4zaysvXtb5CqqltACePiUiFiZf6ZWyRWZLdUl_VUrULIQA72wXFVIbZ4IMKB_eASF9yHkfX6Nv6L1LOeJG_zTPq8JUp3rqk5_HPZuthtTe_2nDdn_U4Yh2RaTDZLefrmBdc4HgS9GTkkNkAVJAs3T8canEpEUNEEf#order-quality)

closed 2024-08-30 by ZefraSK