Weber-State-Submarine-Project/Submarine #83: Rewrite speed sensor code and test accuracy

Ben1219925, 2024-08-30:


ZefraSK, 2024-09-26:
No longer has a weird issue of detecting a 1 instead of a 0. We do not know how to test it yet. It feels a little laggy

ZefraSK, 2024-10-22:
Date: 10/21/24
Current Status:

The Garmin Speed Sensor is not usable. We are not moving nearly fast enough to get the wheel to spin.

Currently I have coding a ping_velocity that calculates velocity using the distance from the ping.

According to a ME, a Pitot Tube will surely suffice to our issue. Pitot Tube simply calculates the velocity based on the pressure of the inside vs outside. I found this one that could our need ([Water Flow Sensor](https://www.pasco.com/products/sensors/pasport/general-flow-sensor)) which will cost around ~$265. This either communicates analog or digital. More research must be done on how it communicates.

ZefraSK, 2024-10-28:
Date: 10/28/24
Current Status:
Ben coded up a xyz position node rather than a velocity node using the ping. We are shoving these values into the EKF for robot_localization. It is somewhat working. 

Currently we need to see if we need a LPF for the sonars

closed 2024-11-21 by ZefraSK